allenporter/pyrainbird #3: Setting for Rain Sensor (Bypass/Active)

lukicsl, 2018-05-29:
Would it be possible to implement setting Rain Sensor setting, e.g. Active/Bypass

ionred, 2018-06-06:
@jbarrancos Curious as well on the API definition -- how did you go about getting that information?

jbarrancos, 2018-06-06:
Hmm not seeing it as part of the default API spec. This would take some nettracing or checking the code. Both will take me some time which i dont’t have right now. I’ll see if i can get some time to analyze the traffic.

jbarrancos, 2018-06-06:
The API is clear text in the android code. Additionally rest is by tracing / decrypting the traffic using a stub service.

lukicsl, 2018-06-07:
Controlling this setting by home automation would make it possible, by actually measuring rain (mm/m2), to precisely override the defined schedule. The sensor HW is a crap, I will remove it.

closed 2019-09-10 by jbarrancos