allenporter/pyrainbird #15: New release seems to only account for 8 zones.

josephkiser04, 2019-10-05:
With release 0.4.1 it seems that data is only pulled for 8 zones. Those of us with more then 8 zones can no longer get information for anything past zone 8. I believe the issue may be in the data.py file States class where there is an iteration 0 through 8. But that is just a complete guess.

`class States(object):
    def __init__(self, mask):
        self.count = len(mask) * 4
        self.mask = int(mask, 16)
        self.states = ()
        rest = mask
        while rest:
            current = int(rest[:2], 16)
            rest = rest[2:]
            for i in range(0, 8):
                self.states = self.states + (bool((1 << i) & current),)`

ironlion27, 2020-04-19:
Any chance we can make that 8

> for i in range(0, 8):

into a 16?

lifeofbrian, 2020-04-20:
The Rainbird devices support up to 22 stations/zones. It may be good to increase to 22 and just indicate in the documentation if it only has been tested for up to 8 zones.

jlinford, 2020-04-26:
Fixed on PR 18 https://github.com/jbarrancos/pyrainbird/pull/18

josephkiser04, 2020-05-27:
@jlinford How can I install and test your pull request here. Doesn't look like this is getting rolled out anytime soon and I would love to get this back up and working. Thanks!

imsorrywhat, 2020-07-08:
> 
> 
> @jlinford How can I install and test your pull request here. Doesn't look like this is getting rolled out anytime soon and I would love to get this back up and working. Thanks!

I tried to load @jlinford changes as a custom component but I didn't have any luck in getting Home Assistant to use it instead of the stock component. I pulled down the rainbird files, made modified the files with the same changes then put it into a rainbird folder in custom_components folder. Maybe you'll have better luck.

jbarrancos, 2020-08-03:
Now pushed to pypi. Someone needs to update the component if needed to latest at HA repo

closed 2022-11-12 by allenporter