allenporter/pyrainbird #16: CombinedControllerStateRequest

michapr, 2020-04-09:
Hi,
thank you for the great lib!

The "CombinedControllerStateRequest" request seems not to work, or I make something wrong.
If I try:
`print(controller.command("CombinedControllerState"))`
get:
```
Exception: Status request failed with wrong response! Requested CC but got 00:
{'type': 'NotAcknowledgeResponse', 'commandEcho': 76, 'NAKCode': 1}
```
I would like to get from anywhere the the remaining time of the sprinkler. Hope to get it from here?

Thanks
Michael
Thanks!

jbarrancos, 2020-08-03:
The json in resources is from the android apk. It is the commands supported at that time by the controller stick. In debug, you should get the details if you fiddle arround a bit. It's been too long for me when I've played with this on my old rainbird setup. I've tried to keep it as simple as possible for anyone to adapt the scripts.

tsfran, 2022-04-16:
Hello. 

I'm new. I've been playing with and enjoying pyrainbird--a highly useful lib!

Has anyone gotten the "CombinedControllerState" to work? 

If so, can you share the 'secrets'?

I've been getting the same issue as michapr and I would really like to be able to get the information from The "CombinedControllerState".

Alternatively, how can I get the seasonal adjustment data (i.e., is there a 'get_water_budget' command)?

Thanks!

allenporter, 2023-01-07:
@tsfran I've added an example in https://github.com/allenporter/pyrainbird/blob/main/examples/mitm_rainbird.py for an mitmproxy plugin that can decode the requests.

allenporter, 2023-01-08:
I don't currently see CombinedControllerState exposed as an API object, i think this should be added and tracked somewhere else. Closing.

closed 2023-01-08 by allenporter