allenporter/pyrainbird #20: Too many warnings in the logs

peternijssen, 2020-05-26:
Within Home Assistant, we noticed that this library is throwing too many warnings in the log files. See https://github.com/home-assistant/core/issues/33614.

The problems seems to be caused by the fact that the controller is not always available or perhaps too busy to handle the request, and thus throws a `503 Service Unavailable` error.

I believe the problem relates to the for loop in this line: https://github.com/jbarrancos/pyrainbird/blob/master/pyrainbird/client.py#L40

Basically when a command fails, we retry it again till we reached the retry threshold. I don't think it's therefor necessary to keep throwing the warning all the time, as the system tries to correct itself. I would suggest to only throw warnings when the retries have been exhausted.

mikesalz, 2020-06-15:
I guess there has not been any traction on this?

mikesalz, 2020-07-19:
@jbarrancos I am encountering this too, as well as many others. It would be great if there was a way to suppress the warnings!

jbarrancos, 2020-07-19:
The logging of warnings can be changed to debug for sake of cleaning up the logging, but there is no warning if all retry’s failed. Which is not perfect either. The other issue is, i don’t have a rainbird installation anymore. So i can’t change and test this. Someone else needs to pick up and test this fix.

allenporter, 2022-11-12:
I see this myself as my rainbird is super close to wifi. I'll have a look

allenporter, 2022-11-12:
Currently an offline device reports these log messages:
```
2022-11-12 09:41:55.506 WARNING (SyncWorker_2) [pyrainbird] Unable to connect: HTTPConnectionPool(host='xxxx, port=80): Max retries exceeded with url: /stick (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f1cee2ed8e0>: Failed to establish a new connection: [Errno 113] No route to host'))
2022-11-12 09:41:55.507 WARNING (SyncWorker_3) [pyrainbird] Unable to connect: HTTPConnectionPool(host='xxxx, port=80): Max retries exceeded with url: /stick (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f1cee2fb430>: Failed to establish a new connection: [Errno 113] No route to host'))
2022-11-12 09:41:55.507 WARNING (SyncWorker_2) [pyrainbird] Response not returned.
2022-11-12 09:41:55.508 WARNING (SyncWorker_3) [pyrainbird] Response not returned.
```

I haven't looked closely, but we may want to update API to return an error instead of logging and let the caller handle logging as an alternative.

allenporter, 2022-12-20:
I've added an asyncio based client library which does not log any warnings/errors and instead throws exceptions. I plan to incorporate this into home assistant. I'm closing this for now given the library has the necessary support, and we won't track downstream fixes here.

closed 2022-12-20 by allenporter