allenporter/pyrainbird #22: Unable to connect to the stick - Failed to parse

licarth, 2020-08-18:
Hi, 

I'm trying to use this library with an ESP-RZXe. I'm getting the following error message when trying to call `controller.irrigate_zone(1.1)` :

```python
from pyrainbird import RainbirdController
controller = RainbirdController("‏192.168.8.174","myPassword")
controller.irrigate_zone(1,1);
```

I am getting:

```
Unable to connect: Failed to parse: http://‏192.168.8.174/stick
Response not returned.
Unable to connect: Failed to parse: http://‏192.168.8.174/stick
Response not returned.
Unable to connect: Failed to parse: http://‏192.168.8.174/stick
Response not returned.
Empty response from controller
```

However, I can control the module from the RainBird Android application.

Anyone experiencing the same symptoms?

Version of the LNK WiFi module software : 1.63.
If required, I could investigate this a bit more and intercept the http communication between the LNK module and my android app (with a dummy password, so that we can decrypt the payload).

holoduke, 2020-10-18:
Same issue here. Is there somekind of API reference guide made by rainbird. so i can fix it myself.

allenporter, 2023-01-05:
You should be able to enable debug logging to see more details about the request. Also see in the `examples` directory there is an mitmproxy add-on for decoding payloads.