allenporter/pyrainbird #24: Several problems compiling, 'requirements.txt' missing at Pypi, installation of 'test' package

onkelbeh, 2020-09-12:
Hello Jordy,

first, I hope you don't mind I included your component for [Home Assistant Gentoo Overlay](https://github.com/onkelbeh/HomeAssistantRepository).

During compile test I saw that your SDIST at Pypi is missing some files, at least 'requirements.txt':
```
* Package:    dev-python/pyrainbird-0.4.1
 * Repository: HomeAssistantRepository
 * Maintainer: b@edevau.net
 * Upstream:   jordy@fusion-ict.nl
 * USE:        abi_x86_64 amd64 elibc_glibc kernel_linux python_targets_python3_8 test userland_GNU
 * FEATURES:   network-sandbox preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
>>> Unpacking pyrainbird-0.4.1.tar.gz to /var/tmp/portage/dev-python/pyrainbird-0.4.1/work
>>> Source unpacked in /var/tmp/portage/dev-python/pyrainbird-0.4.1/work
>>> Preparing source in /var/tmp/portage/dev-python/pyrainbird-0.4.1/work/pyrainbird-0.4.1 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/dev-python/pyrainbird-0.4.1/work/pyrainbird-0.4.1 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/dev-python/pyrainbird-0.4.1/work/pyrainbird-0.4.1 ...
 * python3_8: running distutils-r1_run_phase distutils-r1_python_compile
python3.8 setup.py build -j 10
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    with open("requirements.txt") as fd:
FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
 * ERROR: dev-python/pyrainbird-0.4.1::HomeAssistantRepository failed (compile phase):
 *   (no error message)
```
Gentoo prefers SDIST form Pypi for Python packages, so I tried this first.
As a fallback, I now normally I would use the source from github.com.

This was not possible, because you did not tag releases there.
So I cloned it, and added the missing release tags:

```bash
root@g18-hasstest2:/tmp/pyrainbird # git tag 0.4.1 044e67e1597252420c3b7b46f54a8872f448f419
root@g18-hasstest2:/tmp/pyrainbird # git tag 0.4.2 d9fc7b4a1c845277ef6f0b4501e8053ef0556453
root@g18-hasstest2:/tmp/pyrainbird # git push origin 0.4.1
Gesamt 0 (Delta 0), Wiederverwendet 0 (Delta 0), Pack wiederverwendet 0
To github.com:onkelbeh/pyrainbird.git
 * [new tag]         0.4.1 -> 0.4.1
root@g18-hasstest2:/tmp/pyrainbird # git push origin 0.4.2
Gesamt 0 (Delta 0), Wiederverwendet 0 (Delta 0), Pack wiederverwendet 0
To github.com:onkelbeh/pyrainbird.git
 * [new tag]         0.4.2 -> 0.4.2
```
-> https://github.com/onkelbeh/pyrainbird/releases

During the next compilation from the nox valid sources, my test script threw an error, because `setup.py` tries to install a package called `test` at top level.

I patched this for now at my repo (https://github.com/onkelbeh/HomeAssistantRepository/commit/ad92e3e58121e04185bc626769f4e7acb8d96d37).

I will send a PR for the top level violation in a few seconds.

Thanks a lot.
 \B.

onkelbeh, 2020-09-12:
Also see https://git.edevau.net/onkelbeh/HomeAssistantRepository/issues?type=all&state=open&labels=&milestone=0&assignee=0&q=pyrainbird

allenporter, 2022-11-12:
Tracking resolution in #34

closed 2022-11-12 by allenporter