allenporter/pyrainbird #27: Stops at 16 Zones

phiz118, 2020-12-30:
I currently have 18 zones at my home.  Rainbird controllers can support much more than 16.  It looks like the code was updated to support more than 8, but why not 24 or 32?

idoc69, 2022-09-25:
I was just looking at this myself as our system has cray numbering and the last station is station 17 despite having only 11 zones. If you change line 223 of _ini_.py from
    lambda resp: States((mask % resp["activeStations"])[:4]), to
    lambda resp: States((mask % resp["activeStations"])[:**5**]),
and line 92 of data.py from
    def __init__(self, mask="0000"):
    def __init__(self, mask="0000**0**"):
this fixes the issue on my Home Assistant setup

allenporter, 2023-01-21:
@idoc69  would you be able to enable debug logging and record the response from the server with the active stations in it? I think that would increase our confidence about ability to have tests for this case to ensure it works going forward.

closed 2026-06-13 by allenporter