allenporter/pyrainbird #34: SDIST on Pypi is missing requirements.txt

onkelbeh, 2021-07-16:
Perhaps you should add a MANIFEST.in file.

```
>>> Emerging (1 of 1) dev-python/pyrainbird-0.4.3::HomeAssistantRepository
>>> Failed to emerge dev-python/pyrainbird-0.4.3, Log file:
>>>  '/var/tmp/portage/dev-python/pyrainbird-0.4.3/temp/build.log'
>>> Jobs: 0 of 1 complete, 1 failed                 Load avg: 0.21, 0.23, 0.20
 * Package:    dev-python/pyrainbird-0.4.3
 * Repository: HomeAssistantRepository
 * Maintainer: b@edevau.net
 * Upstream:   jordy@fusion-ict.nl
 * USE:        abi_x86_64 amd64 elibc_glibc kernel_linux python_targets_python3_9 test userland_GNU
 * FEATURES:   network-sandbox preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
>>> Unpacking pyrainbird-0.4.3.tar.gz to /var/tmp/portage/dev-python/pyrainbird-0.4.3/work
>>> Source unpacked in /var/tmp/portage/dev-python/pyrainbird-0.4.3/work
>>> Preparing source in /var/tmp/portage/dev-python/pyrainbird-0.4.3/work/pyrainbird-0.4.3 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/dev-python/pyrainbird-0.4.3/work/pyrainbird-0.4.3 ...
locale: Cannot set LC_ALL to default locale: No such file or directory
>>> Source configured.
>>> Compiling source in /var/tmp/portage/dev-python/pyrainbird-0.4.3/work/pyrainbird-0.4.3 ...
 * python3_9: running distutils-r1_run_phase distutils-r1_python_compile
python3.9 setup.py build -j 6
Traceback (most recent call last):
  File "/var/tmp/portage/dev-python/pyrainbird-0.4.3/work/pyrainbird-0.4.3/setup.py", line 7, in <module>
    with open("requirements.txt") as fd:
FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
 * ERROR: dev-python/pyrainbird-0.4.3::HomeAssistantRepository failed (compile phase):
 *   (no error message)
 *
 * Call stack:
 *     ebuild.sh, line  127:  Called src_compile
 *   environment, line 2829:  Called distutils-r1_src_compile
 *   environment, line 1151:  Called _distutils-r1_run_foreach_impl 'distutils-r1_python_compile'
 *   environment, line  454:  Called python_foreach_impl 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line 2512:  Called multibuild_foreach_variant '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line 2043:  Called _multibuild_run '_python_multibuild_wrapper' 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line 2041:  Called _python_multibuild_wrapper 'distutils-r1_run_phase' 'distutils-r1_python_compile'
 *   environment, line  759:  Called distutils-r1_run_phase 'distutils-r1_python_compile'
 *   environment, line 1142:  Called distutils-r1_python_compile
 *   environment, line 1002:  Called esetup.py 'build' '-j' '6'
 *   environment, line 1578:  Called die
```

konikvranik, 2022-01-16:
I'm confirming this issue:

```
Defaulting to user installation because normal site-packages is not writeable
Collecting pyrainbird
  Downloading pyrainbird-0.5.0.tar.gz (6.6 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3.8 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4fy8lzqo/pyrainbird_c36dd014791b42bb809fb05022e98b9e/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4fy8lzqo/pyrainbird_c36dd014791b42bb809fb05022e98b9e/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-7ozawii7
         cwd: /tmp/pip-install-4fy8lzqo/pyrainbird_c36dd014791b42bb809fb05022e98b9e/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-4fy8lzqo/pyrainbird_c36dd014791b42bb809fb05022e98b9e/setup.py", line 7, in <module>
        with open("requirements.txt") as fd:
    FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/4e/1e/d15c493034097e2b976738ec061ad0a431c0121e02291837636be5b7ed75/pyrainbird-0.5.0.tar.gz#sha256=e85b921af138210982b53e69f4166a3af25461bec8982f4a60836f13e7c78ebe (from https://pypi.org/simple/pyrainbird/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

onkelbeh, 2022-11-13:
Just looked at it, thanks. Will try it when it's bumped at HA.

BTW: the exclude in https://github.com/jbarrancos/pyrainbird/blob/master/setup.cfg ("tests") does not match the actual name of your test directory. This could lead into a top level violation.

allenporter, 2022-11-13:
Will keep open to fix test issue

PR #52 (2022-11-13): Exclude test from pyraindbird package
Exclude test from pyraindbird package

Fixes #34
closed 2022-11-13 by allenporter