allenporter/pyrainbird #38: pyrainbird v0.5.0 is not installable

konikvranik, 2022-01-30:
Trying to upgrade to pyrainbird-0.5.0.tar.gz:

```
> pip install --upgrade pyrainbird

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyrainbird
  Using cached pyrainbird-0.5.0.tar.gz (6.6 kB)
    ERROR: Command errored out with exit status 1:
     command: /srv/homeassistant-p3.9/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1uoc0gk4/pyrainbird/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1uoc0gk4/pyrainbird/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-narz3ibj
         cwd: /tmp/pip-install-1uoc0gk4/pyrainbird/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-1uoc0gk4/pyrainbird/setup.py", line 7, in <module>
        with open("requirements.txt") as fd:
    FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 20.2.4; however, version 21.3.1 is available.
You should consider upgrading via the '/srv/homeassistant-p3.9/bin/python3.9 -m pip install --upgrade pip' command.
```

michapr, 2022-02-25:
in the package for pip install this file is now missing... Hope that author can fix it...
You can use the install from github (clone and call setup.py) - but not sure can it work in this case...

allenporter, 2022-07-02:
I'm happy to volunteer to fix the package, and have it built using the standard python github actions -- assuming the maintainer would agree and approve.

I've sent a PR to setup python tests in a github actions as a first step: https://github.com/jbarrancos/pyrainbird/pull/43

jbarrancos, 2022-11-10:
@allenporter you're still willing to maintain? At the moment I already have approx. 1 year no LNK module. Expecting a new one next year.

allenporter, 2022-11-10:
Yeah, I still have a rainbird and happy to do some of these basics here.

(Generally I'm not super up to speed on all the rainbird protocol details so if you had any more details on how it was reverse engineered that'd be useful, but did get some amount of clarity reading the code. I'm interested in maybe extending a little with a few other things I see in the app i can't see in the library)

allenporter, 2022-11-12:
Duplicate of #34

closed 2022-11-12 by allenporter