allenporter/pyrainbird #39: Exception: Status request failed with wrong response

michapr, 2022-02-28:
Could you please check https://community.home-assistant.io/t/rainbird-ha-integration-error/396946 ?
(there more users could find the answer ;) ...)

I get:
```
Exception: Status request failed with wrong response! Requested 01 but got 00:
{'type': 'NotAcknowledgeResponse', 'commandEcho': 57, 'NAKCode': 4}
```

pyrainbird 0.5.0 itself is working (in command line script)

Thanks!

allenporter, 2023-01-07:
@michapr Are you still seeing this issue/

allenporter, 2023-01-07:
Reading the discussion in the community forum i see the full stack trace:
```
2022-02-26 16:49:07 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1718911544] Status request failed with wrong response! Requested 01 but got 00:
{'type': 'NotAcknowledgeResponse', 'commandEcho': 57, 'NAKCode': 4}
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 209, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 663, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 896, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 700, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 948, in async_turn_on
    await self.hass.async_add_executor_job(ft.partial(self.turn_on, **kwargs))
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/rainbird/switch.py", line 82, in turn_on
    response = self._controller.irrigate_zone(int(self._zone), int(self._duration))
  File "/usr/local/lib/python3.9/site-packages/pyrainbird/__init__.py", line 136, in irrigate_zone
    response = self._process_command(
  File "/usr/local/lib/python3.9/site-packages/pyrainbird/__init__.py", line 202, in _process_command
    response = self.command(cmd, *args)
  File "/usr/local/lib/python3.9/site-packages/pyrainbird/__init__.py", line 188, in command
    raise Exception(
Exception: Status request failed with wrong response! Requested 01 but got 00:
{'type': 'NotAcknowledgeResponse', 'commandEcho': 57, 'NAKCode': 4}
```
It seems like the `NotAcknowledgeResponse` is saying the device is rejecting the request to start irrigation. I have not personally seen that before, but I could imagine changing the error message to be more friendly.

michapr, 2023-01-08:
With latest integration looks like there are no more such errors - have just tried with "fresh" installation.

Thanks for reminder ;)

closed 2023-01-08 by michapr