allenporter/pyrainbird #42: Support for LNK2 WiFi

J0EK3R, 2022-05-12:
Hello and many thanks for this project!

I took this project as base for an implementation for the **[FHEM](https://fhem.de) home automation system** written in **Perl**:
**[fhem-rainbird](https://github.com/J0EK3R/fhem-rainbird)**

Now Rain Bird has released a new WiFi stick **[Rain Bird LNK2 WiFi](https://www.rainbird.com/products/lnk2-enhanced-control)**.

Sadly my implementation can't communicate to the device using "tunnelSip" as methode.
The responses for all commands using "tunnelSip" always are:
_{"error":{"code":-32602,"message":"Invalid parameters"},"id":43,"jsonrpc":"2.0","result":{"text":"NoResult"}}_
Communication using the methodes "getNetworkStatus", "getSettings", "getWifiParams" is working.

Now I'm asking myself if my implementation is faulty or if the LNK2 stick has a different structure of the commands!?

Did someone manage to communicate to a LNK2 WiFi with pyrainbird?

emeins, 2022-05-18:
I tested the irrigation test script with LNK2 successfully.

J0EK3R, 2022-05-18:
Thank you a lot for your quick response!

closed 2022-05-18 by J0EK3R