allenporter/pyrainbird #57: Request to set date & time of RainBird

mantorok1, 2023-01-02:
Hi, is it possible to set the date/time of the RainBird via this package? I can see the requests for getting them but not setting in the list of [sip commands](https://github.com/jbarrancos/pyrainbird/blob/master/pyrainbird/resources/sipcommands.yaml).

The RainBird app on my iPhone allows me to set it so I'm wondering if the command request(s) may not have been discovered yet? If not is there a way to reverse engineer it?

mantorok1, 2023-01-02:
I managed to find the commands for setting the date and time. Here's the details - note that the format is in hexadecimal.
To set time:
```
11hhmmss (where hh is hours, mm is minutes & ss is seconds)
```
To set date:
```
13ddmyyy (where dd is day of month, m is month (1 - C) & yyy is year (remember its in hex so 2023 = 7E7 in hex)
```

If anyone is interested in how I found the commands I used [Charles](https://www.charlesproxy.com) to set up a HTTP proxy and configured my iPhone to use that proxy (in WiFi settings). I then used the RainBird app to change the date and time. Charles captured the requests & response which I then decrypted to find the above commands. The algorithm to decrypt can be found [here](https://github.com/jbarrancos/pyrainbird/blob/master/pyrainbird/encryption.py)

allenporter, 2023-01-05:
@mantorok1 thanks for sharing your method. Based on what you said, i've added a mitmproxy plugin that can decode the requests using the python code: https://github.com/jbarrancos/pyrainbird/pull/59

Are you up to send a PR to add the date/time command?

mantorok1, 2023-01-05:
No worries. I can give I a crack. I'm not that familiar with the pyrainbird source code. Is it just matter of adding the new commands to the `sipcommands.yaml` file?

allenporter, 2023-01-05:
I can do it if you tell me the command format. An example discussion is here https://github.com/jbarrancos/pyrainbird/issues/40

Also the maintainer of this library does not want to maintain it anymore so i've moved to doing development for the official library in https://github.com/allenporter/pyrainbird

mantorok1, 2023-01-05:
I think these are command formats:
```
  SetCurrentTimeRequest:
    command: '11'
    parameterOne: 0
    parameterTwo: 0
    parameterThree: 0
    response: '01'
    length: 4
```
Where `parameterOne` is hours, `parameterTwo` is minutes & `parameterThree` is seconds
```
  SetCurrentDateRequest:
    command: '13'
    parameterOne: 0
    parameterTwo: 0
    parameterThree: 0
    response: '01'
    length: 4
```
Where `parameterOne` is day, `parameterTwo` is month & part of year & `parameterThree` is remainder of year.

The month and year are a bit tricky as the month and year are "mashed" together in the last 2 bytes. It's the same date format that is used in the `CurrentDateResponse` (92)

They both return an `AcknowledgeResponse` (01)

allenporter, 2023-01-11:
I've added this to the library, and here is an example using the command line tool:
```
$ python3 examples/rainbird_tool.py --log-level=debug set_current_date 2023-01-09
DEBUG:pyrainbird.async_client:Request to line: 130917E7
DEBUG:pyrainbird.async_client:Request: {"id": 1673419656.7081575, "jsonrpc": "2.0", "method": "tunnelSip", "params": {"data": "130917E7", "length": 4}}
DEBUG:pyrainbird.async_client:Response: {"jsonrpc": "2.0", "result":{"length":2, "data":"0113"}, "id": 0}
DEBUG:pyrainbird.async_client:Response from line: 0113
DEBUG:pyrainbird.async_client:Response: {'type': 'AcknowledgeResponse', 'commandEcho': 19}
None
$ python3 examples/rainbird_tool.py --log-level=debug get_current_date
DEBUG:pyrainbird.async_client:Request to line: 12
DEBUG:pyrainbird.async_client:Request: {"id": 1673419658.5084407, "jsonrpc": "2.0", "method": "tunnelSip", "params": {"data": "12", "length": 1}}
DEBUG:pyrainbird.async_client:Response: {"jsonrpc": "2.0", "result":{"length":4, "data":"920917E7"}, "id": 0}
DEBUG:pyrainbird.async_client:Response from line: 920917E7
DEBUG:pyrainbird.async_client:Response: {'type': 'CurrentDateResponse', 'day': 9, 'month': 1, 'year': 2023}
2023-01-09
```

closed 2023-01-11 by allenporter