allenporter/pyrainbird #233: Error with openhabian

laserir, 2023-08-09:
Trying to use the rainbird_tool.py under Python 3.9 gives me the following:
```
openhabian@openhabian:~ $ python rainbird_tool.py stop_irrigation
Traceback (most recent call last):
  File "/home/openhabian/rainbird_tool.py", line 107, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/openhabian/rainbird_tool.py", line 101, in main
    result = await args.func(client, **method_args)
  File "/home/openhabian/.local/lib/python3.9/site-packages/pyrainbird/async_client.py", line 319, in stop_irrigation
    await self._process_command(lambda resp: True, "StopIrrigationRequest")
  File "/home/openhabian/.local/lib/python3.9/site-packages/pyrainbird/async_client.py", line 467, in _process_command
    decrypted_data = await self._tunnelSip(
  File "/home/openhabian/.local/lib/python3.9/site-packages/pyrainbird/async_client.py", line 454, in _tunnelSip
    result = await self._local_client.request(
AttributeError: 'tuple' object has no attribute 'request'
```

laserir, 2023-08-10:
Downgraded to 1.1.1, which let me use all my old scripts just fine. Merci @ge0rdi

allenporter, 2023-08-11:
Ah, i see. Yes, this was already fixed in #224 in 3.0.1

closed 2023-08-11 by allenporter

ge0rdi, 2023-08-11:
@allenporter 
The thing is that `3.0.1` requires python `3.10`.
`2.1.1` is latest available for python `3.9`.

It is just unfortunate that `pyrainbird` doesn't work for `3.9` users out of the box.

allenporter, 2023-08-11:
@ge0rdi yes, that's right, dropped support for python3.9 in newer versions. I recommend you embrace newer versions (which can be in the for of virtual environments if not system wide)