allenporter/pyrainbird #270: Understand alternatives to serial numbers for devices with "0"

allenporter, 2023-10-06:
Some devices have a serial number of zero, but what are the other capabilities of these devices? 
- Does updating the firmware to a newer version help?
- Does using the cloud APIs to get a mac address work? 

We can test with some volutneers:

# Setup development environment
```
$ python3 -m venv venv
$ source venv/bin/activate
$ pip3 install pyrainbird==4.0.0
```

# Download rainbird tool
You can either get it from cloning this repo from `examples/` or just download the tool
```
$ curl -OL https://raw.githubusercontent.com/allenporter/pyrainbird/main/examples/rainbird_tool.py
$ ./rainbird_tool.py --help
```

# Test out the APIs
```
$ export RAINBIRD_SERVER=192.168.1.10
$ export RAINBIRD_PASSWORD=mypass
$ /rainbird_tool.py get_wifi_params
mac_address='AA:BB:CC:DD:3E:FF' local_ip_address='1.2.3.4' local_netmask='255.255.0.0' local_gateway='1.1.1.1' rssi=-59 wifi_ssid='example' wifi_password='example' wifi_security='wpa2-aes' ap_timeout_no_lan=20 ap_timeout_idle=20 ap_security='unknown' sick_version='Rain Bird Stick Rev C/1.63'
```

btaczala, 2023-10-06:
Here's mine:
```
$ ./rainbird_tool.py get_model_and_version
model: 0003 (ESP-RZXe), version: 2.9
```
```
$ ./rainbird_tool.py get_wifi_params
mac_address='**:**:**:**:**:**' local_ip_address='0.0.0.0' local_netmask='255.255.255.0' local_gateway='192.168.20.1' rssi=-54 wifi_ssid='********' wifi_password='********' wifi_security='automatic' ap_timeout_no_lan=20 ap_timeout_idle=20 ap_security='none' sick_version='Rain Bird LNK2-0.2.39.0/2.39'      
```
```
$ ./rainbird_tool.py get_serial_number
0
```

allenporter, 2023-10-06:
Awesome, so it sounds like `get_wifi_params` can work to give us a unique id for your device. Thank you for verfifying!

allenporter, 2023-10-06:
@btaczala you may be interested in https://github.com/allenporter/pyrainbird/issues/178 as well -- that issue looks like it is about the same device as you have. They were reporting the existing code can't start zones for that device. If that is true for you, you could help capture the actual responses from the device so we can improve support.

tieskuh, 2023-10-06:
I get the following error running this on a raspberry pi (not running HA). Should I run it from my HA instance or am I missing something?

pi@raspberrypi:~ $ pip3 install pyrainbird==4.0.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyrainbird==4.0.0
  Could not find a version that satisfies the requirement pyrainbird==4.0.0 (from versions: 0.0.9, 0.1.0, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.6, 0.2.0, 0.2.1, 0.3.0, 0.3.1, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.5.0, 0.6.0, 0.6.1, 0.6.2, 0.6.3, 0.7.0, 0.7.1, 1.0.0, 1.1.0, 1.1.1, 2.0.0, 2.0.1, 2.1.0, 2.1.1)
No matching distribution found for pyrainbird==4.0.0

tieskuh, 2023-10-06:
I have a ESP-RZXe and LNK2 indeed...

konikvranik, 2023-10-06:
> Here's mine:
>
> ```
> $ ./rainbird_tool.py get_model_and_version
> model: 0003 (ESP-RZXe), version: 2.9
> ```
>
> ```
> $ ./rainbird_tool.py get_wifi_params
> mac_address='**:**:**:**:**:**' [...]

I have the same RainBird, but different stick:

```
model: 0003 (ESP-RZXe), version: 2.9
mac_address='**:**:**:**:**:**' local_ip_address='*.*.*.*' local_netmask='*.*.*.*' local_gateway='*.*.*.*' rssi=-78 wifi_ssid='***' wifi_password='***' wifi_security='wpa2-aes' ap_timeout_no_lan=20 ap_timeout_idle=20 ap_security='unknown' sick_version='Rain Bird Stick Rev C/1.63'
serial: 0
```

adorobis, 2023-10-06:
Here is mine:
```
(venv) root@f5d520f7c9c4:~# ./rainbird_tool.py get_model_and_version
model: 0006 (ST8x-WiFi), version: 1.0
(venv) root@f5d520f7c9c4:~# ./rainbird_tool.py get_wifi_params
mac_address='4C:A1:61:02:5E:99' local_ip_address='10.144.1.121' local_netmask='255.255.255.0' local_gateway='10.144.1.1' rssi=-59 wifi_ssid='******' wifi_password='******' wifi_security='wpa2-aes' ap_timeout_no_lan=20 ap_timeout_idle=20 ap_security='unknown' sick_version='Rain Bird Stick Rev C/1.63'
```

adorobis, 2023-10-06:
> I get the following error running this on a raspberry pi (not running HA). Should I run it from my HA instance or am I missing something?

You can run it anywhere where you can install a python environment. For me the easiest was just to start a new docker container like that (it didn't work in latest one as it is a python 3.12):
```
docker create -it --name python python:3.11
docker start python
docker exec -it python bash
```
And you'll be in a container's shell where you set up the venv as described in the first post here.

adorobis, 2023-10-06:
> sick_version='Rain Bird LNK2-0.2.39.0/2.39'

@btaczala 
I can see that you have the LNK2 module, is it stable? Does it always connect to the Rainbird mobile app and the HA integration? According to Rainbird this new version should be much better than the old one...

luixal, 2023-10-06:
> Here is mine:
> 
> ```
> (venv) root@f5d520f7c9c4:~# ./rainbird_tool.py get_model_and_version
> model: 0006 (ST8x-WiFi), version: 1.0
> (venv) root@f5d520f7c9c4:~# ./rainbird_tool.py get_wifi_params
> mac_address='4C:A1:61:02:5E:99' local_ip_address='10.144.1.121' local_netmask='255.255.255.0' local_gateway='10.144.1.1' rssi=-59 wifi_ssid='******' wifi_password='******' wifi_security='wpa2-aes' ap_timeout_no_lan=20 ap_timeout_idle=20 ap_security='unknown' sick_version='Rain Bird Stick Rev C/1.63'
> (venv) root@f5d520f7c9c4:~# ./rainbird_tool.py get_serial_number
> 0
> ```

I got this one too and the same stick.

Just in case it hasn't been noticed, @btaczala stick shows `ap_security` as `none` but this stick returns value `unkown` for the same field.

ThomasBergholdWieser, 2023-10-06:
Heres mine:

` (venv) root@ae12c1f172bd:/#  ./rainbird_tool.py get_model_and_version
model: 0003 (ESP-RZXe), version: 2.9`

and (removed hte MAC)

`(venv) root@ae12c1f172bd:/# ./rainbird_tool.py get_wifi_params
mac_address='##:##:##:##:##:##' local_ip_address='0.0.0.0' local_netmask='255.255.255.0' local_gateway='192.168.178.1' rssi=-85 wifi_ssid='MyWifi' wifi_password='********' wifi_security='automatic' ap_timeout_no_lan=20 ap_timeout_idle=20 ap_security='none' sick_version='Rain Bird LNK2-0.2.39.0/2.39'`

btaczala, 2023-10-06:
> @btaczala you may be interested in #178 as well -- that issue looks like it is about the same device as you have. They were reporting the existing code can't start zones for that device. If that is true for you, you could help capture the actual responses from the device so we can improve support.

> > sick_version='Rain Bird LNK2-0.2.39.0/2.39'
> 
> @btaczala I can see that you have the LNK2 module, is it stable? Does it always connect to the Rainbird mobile app and the HA integration? According to Rainbird this new version should be much better than the old one...

Yeah mine works more or less stable. Though more and more I'm considering replacing it with my own ESP32 (with esphome) as a controller. 
HA sometime reports is as unresponsive but the issue is gone after a while (though my lawn wasn't properly watered a time of two because of this).

luixal, 2023-10-06:
I haven't had any availability issues with that adapter. Only thing (on rainbird side I guess) is that it takes a lot of time to load the schedule using rainbird's app.

I bought one of [this zogbee rele board](https://es.aliexpress.com/item/1005005671033060.html?spm=a2g0n.order_detail.order_detail_item.3.1c1f39d3lzEitM&gatewayAdapt=glo2esp)  to replace the controller as I need more zones now, but haven't tested it yet.

allenporter, 2023-10-07:
Thanks all! We now have what we need.

closed 2023-10-07 by allenporter