allenporter/pyrainbird #315: get_zone_states on LXIVM controller

funkzy, 2023-12-07:
Hi,

I have the LXIVM two wire controller and would like to write my own interface for controlling the device. I noticed that out of the box, some of the pyrainbird exposed API's work and some do not. (For reference, the LXIVM uses the LNK2 wifi module.)

For get_zone_states, stations 1-24 work:

# running zone 9
$ python3 rainbird_tool.py get_zone_states
states: 1:0, 2:0, 3:0, 4:0, 5:0, 6:0, 7:0, 8:0, **9:1**, 10:0, 11:0, 12:0, 13:0, 14:0, 15:0, 16:0, 17:0, 18:0, 19:0, 20:0, 21:0, 22:0, 23:0, 24:0

But running zone 27 is not displayed:

$ python3 rainbird_tool.py get_zone_states
states: 1:0, 2:0, 3:0, 4:0, 5:0, 6:0, 7:0, 8:0, 9:0, 10:0, 11:0, 12:0, 13:0, 14:0, 15:0, 16:0, 17:0, 18:0, 19:0, 20:0, 21:0, 22:0, 23:0, 24:0

Any suggestions on how to display > 24 stations?

Thanks!

allenporter, 2023-12-09:
Thanks for the report.

Since i don't have one of these devices, we'll want to capture some of the data here.  Maybe running with `--log-level=debug` will give us enough detail to understand the change needed?

allenporter, 2023-12-09:
Otherwise you can see https://github.com/allenporter/pyrainbird/blob/main/CONTRIBUTING.md for how to start up a proxy to get request traces from the app. See discussion in https://github.com/allenporter/pyrainbird/issues/178

funkzy, 2025-02-17:
Just getting back to this now - the current state of the union is:

- get_zone_states displays the status of stations > 24 (woohoo!)
- I can't irrigate_zone for either zone 23 or zone 28 and strangely they produce different errors:

### Irrigate zone 23:

$ python3 rainbird_tool.py --log-level=debug irrigate_zone 23 1
Traceback (most recent call last):
  File "/Users/x/python_projects/quickbird/rainbird_tool.py", line 109, in <module>
    asyncio.run(main())
    ~~~~~~~~~~~^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/base_events.py", line 720, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/Users/x/python_projects/quickbird/rainbird_tool.py", line 103, in main
    result = await args.func(client, **method_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/async_client.py", line 327, in irrigate_zone
    await self._process_command(
        lambda resp: True, "ManuallyRunStationRequest", zone, minutes
    )
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/async_client.py", line 481, in _process_command
    data = rainbird.encode(command, *args)
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/rainbird.py", line 196, in encode
    return encode_command(command_set, *args)
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/rainbird.py", line 213, in encode_command
    params = (cmd_code,) + tuple(map(lambda x: int(x), args))
                           ~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/rainbird.py", line 213, in <lambda>
    params = (cmd_code,) + tuple(map(lambda x: int(x), args))
                                               ~~~^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'datetime.time'

### Irrigate zone 28:

$ python3 rainbird_tool.py --log-level=debug irrigate_zone 28 1
DEBUG:pyrainbird.async_client:Request (ManuallyRunStationRequest): 39002801
DEBUG:pyrainbird.async_client:Request: {"id": 1739750821.3688939, "jsonrpc": "2.0", "method": "tunnelSip", "params": {"data": "39002801", "length": 4}}
DEBUG:pyrainbird.async_client:Response: '{"id":1739750821.3688939,"jsonrpc":"2.0","result":{"data":"003904","length":3}}'
DEBUG:pyrainbird.async_client:Response from line: 003904
DEBUG:pyrainbird.async_client:Response: {'type': 'NotAcknowledgeResponse', 'commandEcho': 57, 'NAKCode': 4}
ERROR:pyrainbird.async_client:Request (ManuallyRunStationRequest) failed with wrong response! Requested ({'01'}), got 00:
{'type': 'NotAcknowledgeResponse', 'commandEcho': 57, 'NAKCode': 4}
Traceback (most recent call last):
  File "/Users/x/python_projects/quickbird/rainbird_tool.py", line 109, in <module>
    asyncio.run(main())
    ~~~~~~~~~~~^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/base_events.py", line 720, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/Users/x/python_projects/quickbird/rainbird_tool.py", line 103, in main
    result = await args.func(client, **method_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/async_client.py", line 327, in irrigate_zone
    await self._process_command(
        lambda resp: True, "ManuallyRunStationRequest", zone, minutes
    )
  File "/Users/x/python_projects/quickbird/lib/python3.13/site-packages/pyrainbird/async_client.py", line 500, in _process_command
    raise RainbirdApiException("Unexpected response from Rain Bird device")
pyrainbird.exceptions.RainbirdApiException: Unexpected response from Rain Bird device

Any ideas on what might be going on? Thank you!

allenporter, 2025-02-17:
I think this is having a problem because the client library is sending the request incorrectly. I think we need to capture this happening from a real device with mitm proxy to understand how the requests and responses should look since they client is not supporting it right yet.

funkzy, 2025-02-17:
Sounds good - I'll get a capture going.

funkzy, 2025-02-18:
Sent via email

funkzy, 2025-02-21:
Here is the payload sent to run zone 22:

--- Decrypted ---
{"id":1,"jsonrpc":"2.0","method":"tunnelSip","params":{"data":"4B001603","length":4}}
--- request params ---
{'data': '4B001603', 'length': 4}
--- decoded request data ---
{'type': 'StackManuallyRunStationRequest', 'page': 0, 'zone': 22, 'minutes': 3}

And here is the payload sent to run zone 28:

--- Decrypted ---
{"id":1,"jsonrpc":"2.0","method":"tunnelSip","params":{"data":"4B001C03","length":4}}
--- request params ---
{'data': '4B001C03', 'length': 4}
--- decoded request data ---
{'type': 'StackManuallyRunStationRequest', 'page': 0, 'zone': 28, 'minutes': 3}

So it appears that the '16' and '1C' byte is a simple hex encoding of the zone number?

allenporter, 2025-03-03:
Thanks, looks like i can't view the raw dump you sent since the packets are encoded with the device password.  Maybe you can send again the decoded irrigation request if you can find it?

funkzy, 2025-03-03:
Yes, is there a way to decode and save the entire capture?

allenporter, 2025-03-03:
I don't know but there are some videos on the mitm site now https://docs.mitmproxy.org/stable/web-tutorials/web-01-user-interface/

PR #422 (2024-05-10): Add support for full set of 32 states
Remove unnecessary truncation of the zone states, and add test coverage for higher numbered zones.

Fixes #315
commit d274c10a (2024-05-10): Add support for full set of 32 states (#422)

Remove unnecessary truncation of the zone states, and add test coverage
for higher numbered zones.

Fixes #315
closed 2026-06-13 by allenporter